taweili/ardublock #133: I don't import to eclipse

GyuTae, 2015-03-17:
I have problem

I download openblocks-master and ardublock-master.

openblocks-master and ardublock-master import to eclipse.

I must not import.

help me~

catnull, 2015-04-22:
execute "mvn eclipse:eclipse" under the directory of audublock root. then it will work. need to download maven-3.0.x , and  it will be good. and maven-3.2.x, maven-3.3.x don't work well.

GyuTae, 2015-04-24:
thanks!
one more question
my maven version 3.3.1
eclipse import ardublock-master, openblocks-master but maven-install-2.5 file error

catnull, 2015-04-25:
you need to download maven3.0.1 , maven 3.3.1has a repositary access error,as happened when i used try to import to excute mvn.also it needs  newest jdk environment to work well. as i know the arduino and ardu block 's jre /jdk 's version is not up to date  on time.

GyuTae, 2015-04-28:
OK, Thanks!
one more question very sorry!!

arduino IDE -> C:\Users\abu\Documents\Arduino\tools\ArduBlockTool\tool\my file.jar
Not find ArduBlock under the Tool menu

catnull, 2015-07-31:
you should copy the file named "ardublock-all.jar"  compiled and packaged by mvn tool's processing to the directory of "[D:\arduino\workspace\]tools\ArduBlockTool\tool" where the string in [] should  be changed to your own set with Arduino IDE.