taweili/ardublock #138: scoop and inputs

steve24rdg, 2015-08-08:
Can you please help.
I have downloaded and installed ardublock.
It all seems to work except for the blocks under the scoop tab.
each time i try to upload to arduino I get the error that no loop is found.

Do you have any ideas what I am doing wrong?

Also there does not seem to be any blocks to setup pins as input analog or input digital
I see only digital and analog write in pin setup.

Thank you for your time

steve24rdg, 2015-09-02:
Anyone ????

taweili, 2015-09-02:
Could you upload the abp file?

steve24rdg, 2015-09-02:
can you tell me what is the abp file and how I can find it

steve24rdg, 2015-09-03:
I have several files with the suffix abp .
here is the default one
[image]

emil01, 2015-09-06:
@steve24rdg Take a screenshot of your ardublock code and upload it here so someone can review it.

So far this does not sound like a ardublock issue and more like you wrote the code wrong.

steve24rdg, 2015-09-06:
[image]

emil01, 2015-09-06:
can you try by dragging and dropping just the loop block from control. That seems to solve the no loop problem.

steve24rdg, 2015-09-07:
[image]

Now I get an error in the arduino IDE

emil01, 2015-09-07:
Have you installed the scoop libraries?

steve24rdg, 2015-09-07:
[image]

The scoop library is included in my libraries list under contributed libraries.
I think there is a problem with the library though. Can you please tell me step by step how to install the latest scoop libraries for the latest Arduino IDE.

steve24rdg, 2015-10-13:
So I decided to try removing arduino and ardublock from my computer and re installing it.
I followed the instructions for installing ardublock found on the sparkfun website and I re downloaded the ardublock-scoop jar file.
re installed it all and I get exactly the same problems.
Can somebody please help me fix this problem.
Thanks

steve24rdg, 2015-10-13:
I think that the scoop library is either not compatible with the latest arduino IDE or i am downloading the wrong one.
Anyone please show me the correct scoop library to download?