taweili/ardublock #144: Subroutine with variables

jm4r7in, 2016-02-18:
It's possible to make subroutine (for example do_this(), but is it possible to make subroutines which take variables (for example do_this(int with_that) ?
It would be great !
Thanks you.

dimag0g, 2021-03-24:
IMO this only makes sense if local variables are implemented. Copying a value from one global variable into another on a subroutine call makes no sense.

karlTH, 2021-03-24:
In the version i've made there is subroutine with variables since many year :
[https://github.com/taweili/ardublock/issues/191#issuecomment-805381780](https://github.com/taweili/ardublock/issues/191#issuecomment-805381780)

dimag0g, 2021-03-27:
@karlTH can I see the source code implementing variables in subroutines?