taweili/ardublock #148: I am worried that I think the interface is really bad

TonyCallear, 2016-03-27:
I am worried that I think the interface is really bad – confusing to new users. e.g.

- Way too many Block draws and no way to remove them except re-compile. (If you do that, it can be very difficult to stay on the path to contribute to the master branch.)
- Clashing colours battling for attention.
- New blocks that need specific manufacturer hardware placed in “general” draws.
- Illogical arrangement of blocks, draws and their names which are difficult to navigate. Maintained for historic reasons and compatibility with older guides.

**I would like to persuade for a fresh start in the next release. A forum to discuss with more emphasis on “make it right”.**

**Proposal to fund development is really good** but need somewhere to inform/discuss the direction of this development. I don’t want to fund something that would break the core of what I need Ardublock for.

TonyCallear, 2016-03-28:
An example with some attempt at re-organizing block drawers.

http://www.callear.net/arduino/ardublock-highclare-2016-04.jar

Many manufacturer block draws are missing. They should be there. It is just because I need to simply the intereface and remove drawers if I don't have the hardware to use them.

danielfsc, 2016-03-29:
I agree with Tony. Some time ago someone suggest an after compiled
configuration file.
My suggestion is to add a color scheme configuration file and a block
configuration file.

Another annoying thing is the lack of consistency in the colors of the
blocks. For example, if we use green to math operators, we should not use
green in other block even if it is a lighter green.

TonyCallear, 2016-03-29:
A more complete new UI version to take a look at (as if pull request #150  had been applied) here..

[http://www.callear.net/arduino/ardublock-UI-Experiment.jar](url)

I haven't changed any manufacturer Draws, of course.