taweili/ardublock #151: ArduBlock window can close without warning about unsaved work

TonyCallear, 2016-03-29:
Click the big 'X' to close ArduBlock and it just does it - no warning about losing unsaved work.

I've had a look on how to do this but there are many ptions and I just don't know enough Java to pick the correct one.

Hope someone can fix this?

heqichen, 2016-03-30:
×

​
This feature already in ardublock
if you did some change on canvas, ardublock will remind you if you click
the X button.

TonyCallear, 2016-03-31:
Hi,
I am not good at this - but I tried.
I got something which works when ardublock-all.jar is run by itself but it does not work when run from the Arduino IDE.

I have just noticed that... runing Ardublock from the menu in the Arduino IDE and clicking the 'X' -Close button in  Ardublock does **not** close Ardublock - it just hides the window. If do some work in Ardublock, 'close' it and then click  "tools - Ardublock"  again in the Arduino IDE - all your work is still there.

As a silly example... In Main.java set 
`openblocksFrame.setDefaultCloseOperation(JFrame.DO_NOTHING_ON_CLOSE );`
and ardublock-all.jar does not close when run on it's own. Run from the Arduino software and there is no change - it still  'closes'.

TonyCallear, 2017-04-17:
Please see pull request #166

TonyCallear, 2017-10-29:
Perhaps see issue raised on the Arduino github

https://github.com/arduino/Arduino/issues/6869

TonyCallear, 2017-10-30:
If I understand  #6869  correctly, there is no mechanism to interrupt the Arduino IDE  closing from a tool like ArduBlock - and so ArduBlock silently dies and loses any unsaved work.

And no prospect of the Arduino IDE team coding that just for one little tool like ArduBlock either.

AFAICT other ways to lose work without warning are now taken care of??