taweili/ardublock #157: Digital output pin number from variable - bug

dkuku, 2016-07-25:
I've tried to translate blink without delay example to ardublock and when setting first a variable for pin number and later setting the pin as state it doesn't work because ardublock first sets the VARIABLE as 0 before setup function and then in setup first sets the pin number VARIABLE as output( which is 0) and at the end sets the VARIABLE as number  -this is to late

TonyCallear, 2016-07-25:
See issue #149. 

Fixed in this jar? Please comment. 

http://www.callear.net/arduino/ardublock-highclare-2016-04.jar

dkuku, 2016-07-26:
its working ok with this one

dkuku, 2016-07-27:
@TonyCallear Which branch have you used for this build ?
The interface is tidy and it is the only one where digital input with pullup is working (also a bug??)
And I started to translate it to polish and want to build it but every branch I found has the digital input with pullup missing

closed 2016-07-27 by dkuku

TonyCallear, 2016-07-27:
See pull requests #146 or #150. I've been waiting a while for feedback on these requests (not sure if anyone's listening actually?) so you'd have to apply them yourself. 

Or just see https://github.com/TonyCallear/ardublock

Hope this helps.