taweili/ardublock #159: Some changes needed for ArduinoIDE 1.6.12

facchinm, 2016-10-05:
Since there was a huge cleanup of the Editor code during 1.6.11 -> 1.6.12 switch, some extensions do not behave correctly anymore. Ardublock, for example, fails as reported in https://github.com/arduino/Arduino/issues/5454 . Please update the relevant code so the functionality is restored on newer IDEs (reflections may be necessary when "linking" against the newer jars)

allnamesrtaken000, 2016-10-20:
If anyone has any pointers where to start i.e. some details what has changed in the Arduino Editor and subsequently how to make Ardublock compliant that would be appreciated.

facchinm, 2016-10-20:
Here https://github.com/me-no-dev/EspExceptionDecoder/pull/10 you can find  PR I made for another project to fix 1.6.12 compatibility issues. 
You need to fix `processing.app.Editor.setText` function which is no longer a member of Editor class but it's owned by the running Editor window instance. If you need any help we'll be glad to provide it (also via Developer mailing list)

allnamesrtaken000, 2016-10-21:
# pwd
/root/gitsource/ardublock/src/main
# grep -r setText .
./java/com/ardublock/translator/block/Esplora/TFT_Text.java:                   "EsploraTFT.setTextSize("+Height+");"+
./java/com/ardublock/ArduBlockTool.java:                ArduBlockTool.editor.setText(source);

# vi ./java/com/ardublock/ArduBlockTool.java
...
...
        public void didGenerate(String source) {
                ArduBlockTool.editor.setText(source);
                ArduBlockTool.editor.handleExport(false);
        }
...
...

Suggestions would be appreciated...

allnamesrtaken000, 2016-10-21:
Thank you sir.

ParkGoTae, 2016-11-29:
Hello, I have a problem.
ardublock can't upload.
I see the page 
(fix compatibility with Arduino > 1.6.12.)
I want to get pde-1.1.jar and arduino-core.
Thank you.

HonestQiao, 2016-12-11:
use stfc/ardublock:
% mvn clean package -up -U
[WARNING] Command line option -up is deprecated and will be removed in future Maven versions.
[INFO] Scanning for projects...
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Building ArduBlock 2.0.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
Downloading: http://repo1.maven.org/maven2/edu/mit/openblocks/1.0.2-SNAPSHOT/maven-metadata.xml
Downloading: http://repo1.maven.org/maven2/arduino/pde/1.1/pde-1.1.pom
[WARNING] The POM for arduino:pde:jar:1.1 is missing, no dependency information available
Downloading: http://repo1.maven.org/maven2/arduino/arduino-core/1.6.12/arduino-core-1.6.12.pom
[WARNING] The POM for arduino:arduino-core:jar:1.6.12 is missing, no dependency information available
Downloading: http://repo1.maven.org/maven2/arduino/pde/1.1/pde-1.1.jar
Downloading: http://repo1.maven.org/maven2/arduino/arduino-core/1.6.12/arduino-core-1.6.12.jar
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 9.312 s
[INFO] Finished at: 2016-12-11T11:42:37+08:00
[INFO] Final Memory: 12M/309M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project ardublock: Could not resolve dependencies for project org.ardublock:ardublock:jar:2.0.1-SNAPSHOT: The following artifacts could not be resolved: arduino:pde:jar:1.1, arduino:arduino-core:jar:1.6.12: Could not find artifact arduino:pde:jar:1.1 in central (http://repo1.maven.org/maven2) -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

"arduino" is not exists in http://repo1.maven.org/maven2/

gregcorbett, 2016-12-12:
@HonestQiao: Try running ‘mvn validate’ first and then do ‘mvn clean package’. When I was first investigating your comment I had the same problem. But now it works! And I think the only thing I did was run a validate command in-between.

HonestQiao, 2016-12-13:
@gregcorbett Thank your suggest.
I use 'mvn install:install-file' to add the lib files by hand.

ParkGoTae, 2016-12-26:
a problem comes up
here : 
Exception in thread "AWT-EventQueue-0" java.lang.RuntimeException: genusName: DuinoEDU_A0 does not exist.
	at edu.mit.blocks.codeblocks.Block.<init>(Block.java:109)
	at edu.mit.blocks.codeblocks.Block.<init>(Block.java:171)
	at edu.mit.blocks.codeblocks.BlockConnector.linkDefArgument(BlockConnector.java:250)
	at edu.mit.blocks.codeblocks.Block.linkAllDefaultArgs(Block.java:829)
	at edu.mit.blocks.renderable.RenderableBlock.linkDefArgs(RenderableBlock.java:919)
	at edu.mit.blocks.workspace.Page.addBlock(Page.java:644)
	at edu.mit.blocks.workspace.Page.blockDropped(Page.java:587)
	at edu.mit.blocks.renderable.RenderableBlock.stopDragging(RenderableBlock.java:1781)
	at edu.mit.blocks.renderable.RenderableBlock.mouseReleased(RenderableBlock.java:1868)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at edu.mit.blocks.renderable.RenderableBlock.processMouseEvent(RenderableBlock.java:1849)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

ggee, 2017-03-31:
Will a new binary release be made available that works with Arduino 1.8.x ?

closed 2017-04-13 by TonyCallear