taweili/ardublock #168: Support ESP8266?

sabas1080, 2017-07-20:
Hi

Is it project support ESP8266?

Thanks

karlTH, 2017-08-04:
yes i do.
i just work on it.
it's in beta mode

i'm french so the block appear in french but it also work in english.

there is Access point, station mode, you can generate a web page.
the esp send data to the web page and the web page can send data to the esp, all in ajax 
(the block for the web page work like the block of guino)

it work with adafruit.io (send and receive data) and wunderground
i think to make block for the oled ssd1306  

you can see some pictures of the work.
i think it will be available at the beginning of september.

it's a all in one ( arduino, +lib + ardublock esp)
[image]
[image]

leonparente, 2017-08-20:
@karlTH 
Will it have support to MQTT?

karlTH, 2017-08-20:
Hello @leonparente 

I don't plan to support MQTT,

But it use Adafruit_MQTT to connect, send and receive data to https://io.adafruit.com
you have the blocks in the first picture i send (the block receive have change)

Now i work on the oled ssd1306 to display data for the esp.

karlTH, 2017-08-28:
Hi everybody,

I publish my work on the ardublock ESP8266,

here are the screenshot :
https://github.com/karlTH/arduino-esp-ardublock#screenshot-of-ardublock-esp

here are the release :
https://github.com/karlTH/arduino-esp-ardublock/releases

it's an arduino  all in one for windows.
unzip and have FUN

roboticboyer, 2017-11-11:
For Linux it's possible to install only the JAR file of Ardublock with ESP8266 under tool directory:
[https://github.com/karlTH/arduino-esp-ardublock/blob/master/tools/ArduBlockToolESP8266/tool/ardublock-all.jar](https://github.com/karlTH/arduino-esp-ardublock/blob/master/tools/ArduBlockToolESP8266/tool/ardublock-all.jar)

karlTH, 2017-11-11:
yes you can,
and you will need the librarie i used for the block.

i 've fixe many bug, and add many block since last update
so i just update the github.

actually i work to add "emoncms" iot block

ggee, 2018-03-06:
@karlTH What library files and where do I put them?  Also, is there plans to merge your new blocks into this project?

karlTH, 2018-03-06:
Hello @ggee  ou can download : 
 ARDUINO AUGMENTE EXPERIMENTAL here :

http://www.duinoedu.com/arduinoaugmente-default.html

this version  will work under windows. (ardublock still work under inux and mac but you have to manage the librarie and board)
you will have the last version of my work.
In the zip file, the libraries are already installed.

the last block i added are :
NTP, emoncms iot , Oled display with qrcode
and i ve fixed some bug

closed 2025-09-01 by sabas1080