taweili/ardublock #171: Not able to import back abp file.

virenderbhanot, 2017-08-08:
Hi,
I created this program in an earlier version. However, I am not able to Load it now. It loads without error but no blocks are shown. The abp file is attached.  (in .txt extension)
[ardublock-preloaded-behaviour.txt](https://github.com/taweili/ardublock/files/1208856/ardublock-preloaded-behaviour.txt)

koosjr, 2018-10-20:
I have exactly the same problem. Most abp files from the older version (20140702) does not work, but some does.

That said, this new version is nearly exactly what I want. I very much like the version with all the "toys" block stripped out.

koosjr, 2018-10-20:
I figured out why!

It is because of the fact that the "Input pullup" block in this new version is the incorrect shape.

All old progams where the "Input pullup" block was used, will not load.