taweili/ardublock #172: Error fetching path with/ardublock /block/ardublock

JuniorEC, 2017-08-28:
I was trying for the first time in the eclipse and it gave error in the class context, I went to look for the specified way and I did not find it, I tried to change and the error remained, somebody already faced this?

`Exception in thread "main" java.util.MissingResourceException: Can't find bundle for base name com/ardublock/block/ardublock, locale pt_BR
	at java.util.ResourceBundle.throwMissingResourceException(Unknown Source)
	at java.util.ResourceBundle.getBundleImpl(Unknown Source)
	at java.util.ResourceBundle.getBundle(Unknown Source)
	at com.ardublock.core.Context.resetWorksapce(Context.java:122)
	at com.ardublock.core.Context.<init>(Context.java:92)
	at com.ardublock.core.Context.getContext(Context.java:79)
	at com.ardublock.ui.OpenblocksFrame.<init>(OpenblocksFrame.java:71)
	at com.ardublock.Main.startOpenblocksFrame(Main.java:44)
	at com.ardublock.Main.startArdublock(Main.java:38)
	at com.ardublock.Main.main(Main.java:23)
`

karlTH, 2017-08-28:
hello,

seem it miss OpenblocksFrame

look at this :

https://github.com/taweili/ardublock/blob/master/install_openblocks

JuniorEC, 2017-08-29:
OK,

I did the installation and it worked, but this time I returned a mistake.

`ardublock.xml: missing bg.DuinoEDU_infrared_compare.description
Exception in thread "AWT-EventQueue-0" java.lang.Error: Destination component not connected to component tree hierarchy
	at javax.swing.SwingUtilities.convertPoint(Unknown Source)
	at edu.mit.blocks.renderable.RBHighlightHandler.updateBounds(RBHighlightHandler.java:168)
	at edu.mit.blocks.renderable.RBHighlightHandler.access$0(RBHighlightHandler.java:167)
	at edu.mit.blocks.renderable.RBHighlightHandler$1.run(RBHighlightHandler.java:87)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
`

closed 2017-09-13 by JuniorEC