taweili/ardublock #175: Consider extracting block drawers into separate libraries

pilhuhn, 2017-10-08:
Currently all the UI definition etc. lives in one big monolith defined by ardublock.xml and sub-packages in com.ardublock.translator.block.

It would be good if those would be (except for the core ones) be moved into their own packaging unit (jar), that ardublock can read on startup.

This way it would be possible to only load definitions that are needed for a certain project to not clutter the UI. Also vendors could bring in their own packages without the need of modifying the Ardublock core and re-releasing a version of it.

TonyCallear, 2017-10-30:
This has been raised before - and I agree. Unfortunately no one has wanted to take on any part of that task. 

Please see latest release here... if that's any use. Feedback on 'TC' and 'Slim' versions welcome.
https://github.com/taweili/ardublock/releases

set-soft, 2019-07-26:
Hi! what pilhuhn says is the ideal, but a simpler step could be to play with the skins and some scripts. Skins could be done using some kind of "include XXXX", where XXXX is a functionality, i.e. a BlockDrawer. Then we could separate the "BlockDrawer"s for each board. In this way each skin is just a bunch of includes. A script solves the inclusion and generates the usable files. Then we just need a mechanism to select the skin.
BTW: Slim and TC looks good, but I have to really test them.