taweili/ardublock #176: Variables in Functions

civanescu, 2017-10-10:
Can subroutines accept values?

I can define subroutines and access them, but didn't find a way to declare local variables and access the subroutines with values. Do you plan a development in this direction?

Thanks,
Claudiu

karlTH, 2017-10-10:
hello,
it's already done but in duinoedu version
look at
http://www.duinoedu.com/default.html

i've also made a version for the ESP8266
it's still in debugging

civanescu, 2017-10-28:
Well... I have a problem with it. 
The software answere very slow, and can't do "new" file.

I run it on a i7-3820 (multithread), 8GB RAM, Windows 10. My computer is loaded under 10% and RAM under 4GB.
I unzip the software in a folder and run it from that.
I also have the original Arduino in its stadard install folder.
Also the performance is very low in Ardublock!

The error I have is:

> Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
> 	at java.awt.image.DataBufferInt.<init>(DataBufferInt.java:75)
> 	at java.awt.image.Raster.createPackedRaster(Raster.java:467)
> 	at java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
> 	at java.awt.GraphicsConfiguration.createCompatibleImage(GraphicsConfiguration.java:186)
> 	at edu.mit.blocks.codeblocks.rendering.BlockShapeUtil.getBevelImage(BlockShapeUtil.java:178)
> 	at edu.mit.blocks.renderable.RenderableBlock.updateBuffImg(RenderableBlock.java:1387)
> 	at edu.mit.blocks.renderable.RenderableBlock.<init>(RenderableBlock.java:271)
> 	at edu.mit.blocks.renderable.RenderableBlock.<init>(RenderableBlock.java:186)
> 	at edu.mit.blocks.renderable.FactoryRenderableBlock.<init>(FactoryRenderableBlock.java:40)
> 	at edu.mit.blocks.workspace.PageDrawerLoadingUtils.loadBlockDrawerSets(PageDrawerLoadingUtils.java:236)
> 	at edu.mit.blocks.workspace.Workspace.loadWorkspaceFrom(Workspace.java:855)
> 	at edu.mit.blocks.controller.WorkspaceController.loadFreshWorkspace(WorkspaceController.java:415)
> 	at com.ardublock.core.Context.resetWorksapce(Context.java:116)
> 	at com.ardublock.core.Context.<init>(Context.java:82)
> 	at com.ardublock.core.Context.getContext(Context.java:69)
> 	at com.ardublock.ui.OpenblocksFrame.<init>(OpenblocksFrame.java:73)
> 	at com.ardublock.ui.ArduBlockToolFrame.<init>(ArduBlockToolFrame.java:3)
> 	at com.ardublock.ArduBlockTool.init(ArduBlockTool.java:33)
> 	at processing.app.Editor.addTools(Editor.java:891)
> 	at processing.app.Editor.buildToolsMenu(Editor.java:746)
> 	at processing.app.Editor.buildMenuBar(Editor.java:497)
> 	at processing.app.Editor.<init>(Editor.java:251)
> 	at processing.app.Base.handleOpen(Base.java:824)
> 	at processing.app.Base.handleOpen(Base.java:810)
> 	at processing.app.Base.handleOpen(Base.java:806)
> 	at processing.app.Base.handleNew(Base.java:745)
> 	at processing.app.Editor$6.actionPerformed(Editor.java:534)
> 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
> 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
> 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
> 	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
> 	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
> 
>

civanescu, 2017-10-28:
P.S. The standard version from arduino.cc is 1.8.4 ! Maybe it does matter! I'll try to overwrite the folder and keep you informed.

civanescu, 2017-10-28:
**OK! I can confirm, I moved the version of ArduBlock from DuinoEDU to Arduino cc (1.8.4). The app is working ok now, and tested the Ardublock using the standard Blink!

It works fine**

I close the thread. Thank you!

closed 2017-10-28 by civanescu