taweili/ardublock #177: Change size of workspace

karlTH, 2017-11-01:
hello,
i try to change the size of the workspace.
but i don't rrived to find where it is.

if someone have an idea.

Thanks

TonyCallear, 2017-11-01:
The canvas width will reform in width as you add blocks at the right-hand-side.

If you want to change the height see
openblocks\src\main\java\edu\mit\blocks\workspace\Page.java at...
101: public static final int DEFAULT_ABSTRACT_HEIGHT = 3000;

and perhaps openblocks\src\main\java\edu\mit\blocks\workspace\BlockCanvas.java to see how it's used.

Hope that helps
Tony

TonyCallear, 2017-11-01:
Is this a for a special purpose or do you think the BlockCanvas is  too small generally?

karlTH, 2017-11-01:
hello,
I think the bockcanvas is too small when you begin to make big programme.

i've made an esp8266 ardublock and for made the interface of a webpage, some block are need and it's take place.

i've change DEFAULT_ABSTRACT_HEIGHT = 3000;
and for openblock i made : mvn install

it change the content of :  .m2\edu\mit\openblocks\
but the canvas don' t chnage of height
will try again

karlTH, 2017-11-01:
Ok in eclipse i change the referenced librairie of openblock in ardublock

Thanks a lot.

closed 2017-11-01 by karlTH