taweili/ardublock #178: Ultrasonic sensor

Zaurez, 2017-11-24:
I have doubt about whether the pre-defined code of ultrasonic sensor block is correct or not.
because when I uploaded it shows

duration = duration/59;

And i think it should be

distance = duration/59;

if everything is correct according to you help me with simple distance calculation using ultrasonic sensor and ardublock.

karlTH, 2017-11-24:
Hello, Zaurez,
the ultrasonic work fine.
change in the code duration by distance, did it compile.
let as it is does it work fine did it compile.

Zaurez, 2017-11-24:
Yes it did compile
But didn't got any value from the sensor for distance
I have already worked through programming but I want to use ardublock for
teaching purpose so please help me out.

karlTH, 2017-11-24:
the code work well, i use ultrasonic with arudblock.
may be bug with echo trigger( pin), or somewhere else

Zaurez, 2017-11-24:
Ok thanks for helping.

I might need your help in future with ardublock. Let me know if can I ask
for your help.