taweili/ardublock #190: Are this project is dropped?

Arhat109, 2019-12-05:
1. The base openblocks has few pages for blocks. In this - only 'main' page.
Can anybody add multipage modes?

2. The page has problem with page sizes when I opening abp-file from other previous version with big size. The part subroutines are looses few blocks .. it is problems inserting back it. The part blocks are placing at outside the page .. it has a memory leak or not?

3. Will be beatiful to have subroutines with parameters .. please ..

Has anybody help me?

heqichen, 2019-12-05:
Hi @Arhat109 ,because of my job, I have no time on this project. what is your purpose of using ardublock?

Arhat109, 2019-12-06:
Hi, HE Qichen.
I am from Russia, Novosibirsk that I am sorry for my bad English. :)
With my son, we used the Ardublock for teachering Arduino at 2016-2018 years because he was 11-12 years old and was very difficult to learn full stack C/C++.
Now, he have some programs for his robots, it was builded as «Arduino as Lego» on Russian this are written on this: https://community.alexgyver.ru/threads/dlja-detej-kak-lego.1398/
With Ardublock he was takes the 3-d place on the "Robofest-2017" on Russia level competition. Thanks for this program.
My handmade poferfull Arduino Mega2560 are written: https://community.alexgyver.ru/threads/samodelnaja-plata-na-baze-mikrokontrollera-avr-atmel-atmega2560-16au.1454/
And board for it with extended SRAM upto 512kb this: https://community.alexgyver.ru/threads/dispetcher-vneshnej-sram-dlja-avr-mega2560-mega128a-etc.1416/
 
I want to give the Ardublock to teachers in school (where is my son) and to show as it works for teachering childrens in this years old (as right? :) )
P.S.
I work as full stack PHP-programmer and don’t know the Java absolutely.
Help me please.. this is really powerfull work, thanks. The BIG thanks! :)

> 
>Hi  @Arhat109 ,because of my job, I have no time on this project. what is your purpose of using ardublock?

vedes, 2020-05-20:
Hi Arhat109. I am also from Russia and lead a circle of robotics for children. I modified the Ardublock a bit to work with children. It may be useful to you. My repository is https://github.com/vedes/ardublock

Привет, Arhat109. Я тоже из России и веду кружок робототехники для детей. Я немного дорабатывал Ардублок для работы с детьми. Может быть это будет вам полезно. Мой репозиторий https://github.com/vedes/ardublock

karlTH, 2020-05-20:
hello, it's not dropped.
i just discover how to make release in github.
so here it is :
[https://github.com/karlTH/ardublock/releases](https://github.com/karlTH/ardublock/releases)

Arhat109, 2020-05-20:
Спасибо, посмотрю обязательно. В чем заключалась доработка? Бегло почитал описание и не увидел, буду признателен если расскажете письмом.

> Hi Arhat109. I am also from Russia and lead a circle of robotics for children. I modified the Ardublock a bit to work with children. It may be useful to you. My repository is [...]

karlTH, 2020-05-20:
> Спасибо, посмотрю обязательно. В чем заключалась доработка? Бегло почитал описание и не увидел, буду признателен если расскажете письмом.  
>     -- С уважением, Владимир.

Возможность задания кириллических Сделана названий Â переменных Ø константах
Â главном меню сделаны кнопки масштаба A
signalé блоки для двухколёсного шасси, подключенного через L298P bouclier
Исправлен перевод на russe
Сделана упрощённая раскладка меню с блоками для детей обучения.
Для установки распакуйте архив tools.zip в папку, где установлено Arduino

cf : [https://github.com/vedes/ardublock/releases](https://github.com/vedes/ardublock/releases)

Arhat109, 2020-05-21:
Thank you!

> 
> 
>Спасибо, посмотрю обязательно. В чем заключалась доработка? Бегло почитал описание и не увидел, буду признателен если расскажете письмом.  
>    -- С уважением, Владимир.
Возможность задания кириллических Сделана названий Â переменных Ø константах
Â главном меню сделаны кнопки масштаба A
signalé блоки для двухколёсного шасси, подключенного через L298P bouclier
Исправлен перевод на russe
Сделана упрощённая раскладка меню с блоками для детей обучения.
Для установки распакуйте архив tools.zip в папку, где установлено Arduino
cf :  https://github.com/vedes/ardublock/releases
—
 
С уважением, Владимир.