taweili/ardublock #191: Project is continued: https://github.com/karlTH/ardublock/releases

HHHartmann, 2020-05-22:
https://github.com/karlTH/ardublock/releases

dimag0g, 2021-03-24:
It seems the most active fork is https://github.com/stfc/ardublock
At least its master branch is 125 commits ahead of the parent repo
Your master branch is 51 commits behind, could you explain how that's an improvement?

karlTH, 2021-03-24:
here the last update 21 march 2021 : [link](http://duinoedu.com/dl/logiciels/arduino/arduino_augmente/version_duinoedu/DERNIERE_VERSION/2_EXPERIMENTAL_VERSION/)
here a sample of what you can do with this ardublock : [link](https://www.youtube.com/watch?v=dKscZlQ3Pb4)

[image]

udif, 2021-03-24:
As far as I remember, when you fork a project, you don't get an independent 'issues' tab on your fork, but you can explicitly enable it on your project settings.

dimag0g, 2021-03-27:
@udif I would expect to see this tab enabled if a project is forked in order to carry on development (as here). Sure you don't need issues tracking if you fork it only to contribute PRs.

dimag0g, 2021-04-05:
@heqichen and @haobug Sorry to disturb you guys, but I don't seem to find the source code of this Ardublock fork. The latest release is from March 2021, and the latest commit on GitHub is from November 2017. Any idea where I could get the missing commits?

heqichen, 2021-04-09:
> @heqichen and @haobug Sorry to disturb you guys, but I don't seem to find the source code of this Ardublock fork. The latest release is from March 2021, and the latest commit on GitHub is from November 2017. Any idea where I could get the missing commits?

Hi @dimag0g, I didn't see the commit after 2020. Maybe you could see the forked repo of @karlTH. 
My last commit is about 6 years ago that refactoring the UI and removing commercial blocks from the ArduBlock.